samartzidis/WinAppleKey #1: Can I use this driver for installed macbook keyboards?

hismethod, 2017-01-09:
Probably most Macbook pro users want to use a fn key for standalone.
but there is never way until now.
Can I use for installed macbook's keyboards?
If not, have you plan to dev for wired or installed keyboard?
Also you have no plan, Can you give me a advise or hint to modify your driver to apply wired keyboard.
Thanks for you share.

samartzidis, 2017-01-09:
In its current form, you can only use it for the bluetooth A1314 and A1644 keyboards. It could get modified I believe to support the full-sized USB Apple keyboard by handling the URB packets coming from the bus driver. I don't know much about the hardware configuration of the MacBook internal keyboard, so I am afraid I cannot give an answer on this one.

hismethod, 2017-01-10:
Thanks for your kindness advise.

closed 2017-01-16 by samartzidis