samartzidis/WinAppleKey #2: Change key mappings

616b2f, 2017-01-11:
Hi, tank you for creating such amazing thing! i love it 😄 

What i miss is a way to setup custom mappings and get media keys to work.
Have you plans to implement such a feature?

Thank you in advance.

Alexej

samartzidis, 2017-01-11:
Hi Alexej,  Glad you find it useful. Yes it's true that I have not done the media keys and any custom mappings functionality yet. It is in my plans to do so as soon as I get some spare time. As a workaround for the media keys or custom mappings you could use other 3rd party apps such as AutoHotKey. Thanks!

samartzidis, 2017-01-15:
PS: To get multimedia keys mapping using AutoHotKey (under the latest release of WinAppleKey), just create and load this AutoHotKey script:

F19::send {Media_Prev}
F20::send {Media_Play_Pause}
F21::send {Media_Next}
F22::send {Volume_Mute}
F23::send {Volume_Down}
F24::send {Volume_Up}

616b2f, 2017-01-15:
Thank you @samartzidis i will try it

closed 2017-01-16 by samartzidis