samartzidis/WinAppleKey #5: I've installed the driver but nothing changes?

jbukuts, 2017-02-20:
I built and ran the Setup.msi and it shows in the Control Panel yet none of the new key bindings work. Is there a step I am missing? The only driver I have installed for the keyboard right now is the base one when I first connected it via bluetooth.

RailRanger, 2017-02-25:
I still had to swap alt and the windows key with sharpkeys. Also you have to use the keyboard in wireless mode.

samartzidis, 2017-02-25:
@jbukuts if -none- of the bindings work and the keyboard is not connected to the usb cable, that is, it is functioning in bluetooth mode, then it could actually be a problem with the driver. 

I have attached a file that you can use to help with this if you want. You can run the .js file with "cscript.exe listHid.js" in an administrative command prompt and *e-mail* me the results.

[listHid.zip](https://github.com/samartzidis/WinAppleKey/files/801238/listHid.zip)

closed 2017-02-25 by samartzidis