samartzidis/WinAppleKey #6: install

scorpio2k2, 2017-03-29:
Hi,
I compiled the exe and installed it but nothing happens.. 
no keymapping

samartzidis, 2017-03-29:
Hi, I have heard of that issue from a couple of other users here with the A1644 (new) model in wireless setup. I am investigating this.

scorpio2k2, 2017-03-30:
oh.. I'm waiting and I'm trying to investigate myself.

scorpio2k2, 2017-03-30:
hi,
Question: can you share a compiled version to see if it works? 
Thanks

samartzidis, 2017-03-30:
The compiled version is already offered in the project page. If you click on "releases" you will find it. I doubt that there is a problem with your own compiled version though. There most likely is an issue with my code that needs fixing.

samartzidis, 2017-03-30:
Hi Valentin,

There already is a compiled version of the software if you click on
"releases" on the main page.

Thank you.

scorpio2k2, 2017-03-30:
The issue was on my side. It works with the file from "releases".

Thank you

closed 2017-03-30 by scorpio2k2

samartzidis, 2017-03-31:
That's good news! Thanks for letting me know.