samartzidis/WinAppleKey #7: Ctrl + LeftShift + T doesn't work.

Eterad, 2017-04-06:
Hi,

When trying to reopen closed tab using LeftShift doesn't work. It does work however with RightShift.

Best,
Adam

Hubro, 2017-04-18:
I just noticed this too. What a weird bug :)

simplecoder, 2017-07-20:
Yes, I noticed this as well. Unfortunately it is a commonly used key combination too.

vogler, 2018-01-29:
Is this still the case? For me it works fine (A1314 and A1644).

WireCrazii, 2018-03-15:
@vogler Yes, this bug still exists in both A1314 and A1644(for me).

samartzidis, 2018-07-01:
This seems to be a hardware issue with the Apple keyboard and not a driver fault. 

Even without the driver installed when you press the (real)Fn+LShift+T keys simultaneously, the keyboard generates a Keyboard Rollover Error as if you're basically pressing more than 5 keys simultaneously, which is that the keyboard hardware can handle (which is weird as it only happens with that specific 3-key combination). I am closing this for now, unless Apple decides to look into this maybe? :}

The alternative as mentioned elsewhere is using the right shift instead.

closed 2018-07-01 by samartzidis