samartzidis/WinAppleKey #9: Sign with proper sertificate

Hubro, 2017-04-06:
At first I thought the lack of a proper certificate was a small nuisance, since all I had to deal with was the small text saying "Testing" on the desktop. After a few hours of use I noticed that testing mode triggers all kinds of anti cheat engines, and completely bars me from playing any online games. In other words I unfortunately can't use this driver until it can be installed without testing mode.

Can anything be done about this? Is it possibly to explicitly trust your certificate and allow the driver to be installed? Do you have any plans regarding this?

samartzidis, 2017-04-06:
Hi Tomas. Sorry to hear this. I was not aware of that potential side-effect
of Test Mode. What I can imagine that is happening is that certain games
may potentially be looking for Test Mode being enabled and not letting you
play them, as this could potentially mean that you have a cheating driver
installed on your machine or that you have fiddled with Windows in such a
way that you could be cheating against the game engine.

Unfortunately it costs a lot of money to get a driver signing certificate
and I even doubt that Microsoft would ever allow a driver produced by a
project such as this, to be signed. But maybe I can have a second look on
the issue.

Out of curiosity, which game/software is complaining about the Test Mode?

Thanks,
George.

Hubro, 2017-04-06:
In this case it was PLAYERUNKNOWN'S BATTLEGROUNDS. More specifically, it was [Battleye](https://www.battleye.com/) that refused to launch in Test Mode, I have no doubt that Blizzard and Valve's (VAC) anti cheat also refuses to launch in Test Mode.

But even just Battleye involves a lot of popular games like Arma 2, Arma 3, DayZ, H1Z1 etc.

What does it cost to get a driver signing certificate? Is it a one-time cost, or a subscription? Is it perhaps a possibility to get somebody with an existing certificate to sign your driver?

RailRanger, 2017-04-16:
I know it's been done before here. http://www.overclock.net/t/1597441/digitally-signed-sweetlow-1000hz-mouse-driver/0_100

I'm willing to throw in $100 but I think it costs like $350. So a kickstarter or something might work. I know a few games that don't work, Rainbow six seige, For Honor, ESEA in CS:GO. It does not affect any of the games I personally care about though.

samartzidis, 2017-04-17:
@RailRanger thanks. That's an interesting link. I always considered driver signing by individuals an almost impossible act, especially after the latest Windows restrictions came in. I will need to do some further research on this.

Hubro, 2017-04-17:
I will chip in as well, if you launch a crowd funding campaign. 350 USD definitely sounds surmountable.

Hubro, 2017-05-13:
Any progress on this? My Apple keyboard has seen very little use lately, due to the extra effort of having to reboot the PC.

Beej126, 2017-06-15:
we may want to try this: https://github.com/hfiref0x/DSEFix

stereokai, 2017-09-22:
Any update on this? :)

Hubro, 2017-09-22:
My awesome wireless keyboard has been collecting dust for months now, and it feels like such a waste when this perfect solution is just sitting here. All it needs is some money thrown at it to make Windows accept it. Are you planning on starting a crowd funding campaign?

Adwerk, 2018-01-30:
+1 for getting it signed e.g. via a gofundme campaign. I realized it would require some extra work on your end, and quite frankly I wouldnt find it amiss if you added a lil on top for your self. I've enjoyed your keyboard driver for year and would probably have a deform hand had it not been for your work, which afaik is the only working solution. I have no idea how many use/need this but I would assume enough, and running testsigning will be a showstopper for many.

If nothing else, thanks for the work you put into it. 

Cheers.

stereokai, 2018-01-30:
Let's make a plan. If we want to get this driver signed, what are the steps?

1. GoFundMe/Kickstarter campaign
2. ... what else?

Let's work it out together! @samartzidis

Adwerk, 2018-01-30:
TBH I'm not entirely sure what it would take to get this signed (other than money). As Samartzidis mentioned above it's not even certain Microsoft would sign it given the nature of the project. If however the "only" roadblock is with the money required to do so, then I believe people are willing to drop come coin to see it though.

zzuper, 2018-01-31:
Juste give me your damn paypal bro, ill pay for the whole thing. Lets get this done asap!

andersforsell, 2018-03-18:
This is so good, pls consider crowdfunding a proper certificate @samartzidis

elgydium, 2018-03-29:
Do we have any news? I'll definitely pitch in. I just want this shit **done**!

**Ela re Giorgara! Na kanoume to proto bima!**

samartzidis, 2018-03-29:
Hello everyone. It is really exciting seeing how much interest there is in this project as well as the willingness to help! Many thanks to all of you! Unfortunately as some of you may already know, the main barrier here is not the actual certificate costs, but the complicated driver signing process required by Microsoft; that is tailored to businesses and hardware vendors rather than to individuals or to open source projects. Nevertheless, given the interest that I see, I will attempt to contact Microsoft and get to know if there are any chances at all. I will let you know as soon as I know something more. Thank you!!

Hubro, 2018-03-29:
I have to say, it's really infuriating how difficult Microsoft is making it to use open source drivers. It really shouldn't be any more difficult than installing/accepting the author's certificate prior to installation.

elgydium, 2018-03-29:
I know for one that I keep buying magic keyboards only to sell them off again. I think this is my 3rd attempt and I'm not backing off this time. Something must be done, there's always a way!

**Many thanks to @samartzidis!**

stereokai, 2018-03-30:
Nice work, @samartzidis! We stand behind you!

jonnyasmar, 2018-06-07:
@samartzidis Willing to donate to this as well if you get a Kickstarter going! Been using WinA1314 for years, but broke my old Apple keyboard and it's not compatible with the new bluetooth version :(

samartzidis, 2018-07-01:
I have extensively looked into the issue. Signing a driver for Windows 10 x64 onwards requires an EV Driver Signing Certificate. These certificates are not just costly, around 450$/year at DigiCert for example, but they can also only be issued to listed companies (not individuals). Microsoft seems to only accept such certificates issued from Symantec, GlobalSign, DigiCert and Entrust at the moment. Further than that, even if you setup a startup company (say) and buy the EV certificate, you also need to pass a driver artifact validation procedure through the Microsoft Hardware Development Portal, which based on their own criteria might reject your software. Microsoft has deliberately made it extremely difficult to discourage individuals from writing and distributing drivers for their operating systems (and to some extend that is understandable). 

I know how many of you are fond of this driver and thank you for your support. Given the situation though, I will keep on occasionally maintaining WinAppleKey for the ones who are happy to run Windows in Test Mode. In the meantime, as I am also keen on Apple hardware (but as a Windows and Linux user), I have started thinking and kinda working at spare time on alternatives such as an embedded device (I am currently experimenting with Arduino Nano and RPI Zero) that will basically allow Apple Hardware to interface into Windows as conventional HID devices. I will let you know once I have something working.

Thanks!

elgydium, 2018-07-01:
I’m very fond of arduino and I think that might actually work given the
fact that arduino is a very capable and efficient electronic prototyping
platform.

Go for it I say!

stereokai, 2018-07-02:
@samartzidis your dedication and investment are remarkable! Thank you!

RailRanger, 2018-07-10:
Thanks a lot for looking into this, that sounds really cool.
In the meantime you can also install the keyboard onto another computer and use the program synergy to control your main computer's keyboard through it. I can play games like PUBG now which require windows test mode to be off.

elgydium, 2018-07-10:
I see you're a man of culture as well. FFP or TPP? What's your tag?

samartzidis, 2018-08-19:
> there is this now: https://magicutilities.net/
>
> mandatory renewal every 2 years feels a bit heavy handed but he put in the
> work, he deserves whatever the market will bear

Interesting effort! Does it support swapping Ctrl/Fn though?

elgydium, 2018-08-19:
*nope. **unfortunately*

samartzidis, 2018-08-19:
If you are a little bit more patient, just to let you know that I am close (a matter of days) to releasing a radically different solution that will be functioning in the form of a USB dongle (based on a Raspberry Pi Zero board) that you will plug into your PC (Windows 95 (!) to Windows 10 and up, Linux) or even into your TV, game console, etc. (basically anything that supports USB HID devices) and will give it access to your favourite Apple keyboard. 

The **only requirement** is that you will need to purchase a Raspberry Pi Zero board, currently costing £9 in the UK and burn this bootable image to its SD card (you can do this from Windows). Other than that, you have something fully plug-and-play with no driver or other software installations needed. You just plug this into your PC's USB port and the keyboard becomes automatically detected (as a HID keyboard plug-and-play device) and usable by Windows. 

For the technically curious, the SD card boots a minified customised Linux kernel in read-only mode, you can just plug/unplug it anytime with no worries about corrupting the underlying filesystem. The device also boots/initialises in about 7 seconds after you plug it into the PCs USB port, which is pretty quick.

The major advantage of this solution, apart from supporting a wide range of operating systems (and not just the latest Windows 10) is that it also works without an operating system at all, such as in circumstances where you want to get into your PC BIOS before you even get the chance to boot into Windows. It also supports the Apple keyboard's multimedia keys out-of-the-box and of course the option to swap the Ctrl-Fn keys.

I have attached photos of my current working prototype along with a screenshot of the current state of the configuration software (implemented in React/Material-UI), which runs from within your browser (no software installation is needed for this as well).

[RaspiKey Prototype](https://file.io/UbtH9N)

So technically, the dongle offers a local mini HTTP server and web-page where you just connect from your PC's browser to configure it. You do not need to install any software on the PC. The mini HTTP server is implemented through Bonjour (Windows) and Linux Avahi (exploiting Apple's Zeroconf architecture). So for the curious, the WiFi hardware is locked down on the Pi along with any generic networking services.

Any comments are very welcome, especially in regards to the requirement to purchase a Pi board. 

If anyone would be interested in receiving a free beta copy of the bootable image and trying it on their Raspberry Pi Zero W, would be of **huge** help. This is a spare time project and it is kinda difficult for me to invest much time on it as I am very busy at my full-time job. Nevertheless I will do my best to get this out ASAP.

Thank you!

RailRanger, 2018-08-19:
That sounds amazing, I'd love to test, but are you using a USB-A addon board? Something like this? https://www.amazon.com/MakerFocus-Raspberry-Required-Provide-Connector/dp/B077W69CD1

Beej126, 2018-08-20:
very cool George!
yes please share links to the bare hardware required

my main interest would be the [latest magic keyboard with numpad](https://www.apple.com/shop/product/MRMH2LL/A/magic-keyboard-with-numeric-keypad-us-english-space-gray), would there be potential for that one as well? the bare minimum keys requiring remapping on that one are:
* swapping <kbd>option</kbd> <=> <kbd>command</kbd>
* <kbd>fn</kbd> => <kbd>ins</kbd>
* <kbd>eject</kbd> => <kbd>fn</kbd>
* swapping <kbd>F1</kbd>...<kbd>F19</kbd> to std function keys by default
* and possibly numpad <kbd>clear</kbd> => <kbd>numlock</kbd> (but that seems to work by default, at least on the old wired full size equivalent)

[image]

samartzidis, 2018-08-20:
Hi @RailRanger,

Yes in this photo I am using a soldered USB-A add-on just to make it look cooler, but you don't reely need any of that!

All you need is a standard micro-USB cable that on the one end you connect to any free USB board on your computer and on the other end you connect it to the micro-USB port of the Pi (the one in the middle) as seen [here](https://blog.gbaman.info/wp-content/uploads/2015/12/IMG_5140-1038x576.jpg)

The Apple keyboard(s) connect through Bluetooth (no cables here!) to the Pi. The purpose of the Pi board is to basically accept any (exotic) Apple keyboard through Bluetooth perform the required translation internally and present it to the Windows OS as if it is a standard USB keyboard.

Thanks,
George.

samartzidis, 2018-08-20:
Hi @Beej126,

I think handling the latest Magic Keyboard would be possible in the same way it is for the A1314 and the A1644 but I would need to get my hands on one first. I will see what I can do. The prototype has currently only been tested with the 2 above keyboards.

In regards to the hardware all you need is **(1) the Pi Zero board** and a **(2) decent micro SD card** of 2GB or more. You also need a plain micro-USB cable (I believe it is a pretty common thing to mention as a hardware requirement here).

The Pi Hut sells the Raspberry Pi Zero W here: https://thepihut.com/products/raspberry-pi-zero-w for £9.30 GBP but depending on your country, you may be able to find it cheaper elsewhere (e.g. on eBay or Amazon). Just bear in mind that you need the simplest and cheapest version really, you don't need a kit bundle or the one with the pre-soldered header.

Thanks.

samartzidis, 2018-08-20:
@Beej126, @RailRanger I will update the thread with download links once they're ready. Thanks!

stereokai, 2018-08-20:
Wow, that is just amazing, @samartzidis!

What if it was possible to connect via bluetooth to the PC as well?

samartzidis, 2018-08-20:
Hi, @stereokai,

That would n't have been a bad concept but the embedded Pi Bluetooth module can either function as a master or as a slave, not as both. Also, you would still need a way to power the Pi board from somewhere. So I thought it's best to combine the power requirement and the communication functionality over a single USB connection to the PC. The USB connection also does not need any setup on the PC where you connect it to, you just connect it and it magically becomes a keyboard device.

Thanks.

samartzidis, 2018-08-21:
Hi @RailRanger. I am currently working on some final touches but it should n't take too long! I will let you know once it's ready so that I can send you a copy of the SD-card image. So that you know I will currently only support the A1314 and the A1644 keyboards. Thanks a lot!

samartzidis, 2018-08-24:
Not that much relevant to this thread but after looking at the aforementioned **magicutilities.net** commercial product, it seems that they are basically re-bundling Apple's own drivers and selling them as their own. Nevertheless, doing this you can't provide a full remapping solution.

_[Strings]
Manufacturer="Apple Inc."
Provider="Magic Utilities Pty Ltd"
Website="http://magickeyboard2.com"
Drv.Name="MagicKeyboard"
..._

samartzidis, 2018-08-28:
After a few months of exhaustively hard work, I am at the pleasant position to announce that RaspiKey.1.0.1-beta has been released!!! [Here](https://github.com/samartzidis/RaspiKey/blob/master/screenshot1.png) is a screenshot of how it currently looks like. You only need a "Raspberry Pi Zero W" device. There is no software or driver installation needed for the host computer, just a Web browser is enough (which, is also, only required when you pair a keyboard to the device for the first time).

The installation is relatively simple:
1. Download [Apple's Bonjour](https://developer.apple.com/bonjour/) and install it to Windows (in Linux you don't need this). I have made the latest 32-bit and 64-bit versions available [here](https://github.com/samartzidis/RaspiKey/tree/master/utilities).
2. Download the latest RaspiKey image from https://github.com/samartzidis/RaspiKey/releases
3. Download [Etcher]([https://etcher.io/]
4. Use Etcher on Windows to burn the RaspiKey image (.xz file) to a 2GB (if bigger doesn't matter) micro-SD card.
5. Insert the micro-SD card to the "Raspberry Pi Zero W" and then connect it to the computer using just the middle micro-USB port (labelled "USB" with tiny white letters). The Raspberry Pi will both draw power from this port as well as use it to communicate with the PC.
6. Wait about 20 seconds. Then open Chrome on Windows at: **http://raspikey.local**
    You should be able to see the RaspiKey configuration page.

RaspiKey currently supports the Apple Wireless Keyboard (A1314) and the Apple Magic Keyboard (A1644). After you pair the keyboard to RaspiKey, do not forget to click the SAVE button. After this, the dongle (and the keyboard) should theoretically be usable on a wide range of devices including any version of Windows (presumably after Windows 98), Linux, Smart TVs, X-Box and PlayStation game consoles etc. You should also be able to use the keyboard under BIOS before booting Windows, but you may need a powered USB hub for this as the BIOS will most likely start faster than RaspiKey will be able to get into user mode. The multimedia keys should work fine. In contrast to the WinAppleKey driver, Windows do not need to be tweaked in Test mode.

Please bare in mind that this is the first release of a beta version. I have been very careful to make it as solid and stable as possible but if in case you notice any bugs please let me know by e-mail. Also please let me know by e-mail of any suggestions that you may have or any requests or ideas for improvement. 

You may notice that RaspiKey will need a device specific license key to function after a certain period of time, e-mail me and I will send you one.

Thanks!

RailRanger, 2018-08-28:
:D it works great! Thanks for your hard work!

That was seriously easy to setup and I can't detect any lag so far, but I have only used it for a few minutes. Is there a way to swap the alt and windows keys?

samartzidis, 2018-08-28:
@RailRanger, you 're welcome! I As it concerns the swap of alt-cmd it is not configurable yet but I will do this next thing. My plan is to basically offer a user configurable key map where you'll be able to do whatever you want with the keys.

samartzidis, 2018-08-28:
PS: I guess the next thing should also be support for the new full-sized Magic Keyboard as a few have mentioned.

Adwerk, 2018-08-28:
You deserve mad props for getting this done, I know it's been no small task. Ordered a PI zero, will hopefully arrive soon.
Adding support for the numeric version would be nice, thought it would tempt me to buy yet another keyboard ... at which point my family most certainly will make me see a psychiatrist =D

samartzidis, 2018-08-28:
@Adwerk that was funny :) Btw, I hope you made sure it's the Pi Zero **W** and not the plain one (or they might send you there as well due to buying so many Pi ;). Cheers!

Adwerk, 2018-08-28:
Aye, bought the W. Are you using a charger or just powering though the USB? I remember my PI 3 giving undervoltage warnings without a dedicated charger, but it had to power a mouse and keyboard as well.

samartzidis, 2018-08-28:
No there is no problem with the supplied usb power whatsoever. Unneeded Pi
devices (e.g. the wifi) are turned off to save power and the CPU footprint
of the RaspiKey software is extremely minimal.

stereokai, 2018-08-29:
Amazing work @samartzidis 

Now just need to get an Apple keyboard 🤣

joowani, 2018-09-15:
this. is. amazing. 

Thank you so much you literally changed my life. 

I use the magic keyboard over wired USB connection (hassle to charge batteries) and I still prefer the driver despite the signature issue (seems more down to metal and cleaner). 

Do you have a bitcoin/ethereum/usdt address? I'd love to donate for your hard work so far and in hope that you continue to maintain the driver solution in the future.

samartzidis, 2018-09-15:
Hi @joowani, thanks for your nice comments! There's a PayPal donate button at the bottom of the main page. I think that the RaspberryPi solution (which you can view as a specialized dongle) has a bigger potential and applicability compared to the pure Windows driver because it makes the keyboard usable by any device that has a USB port and requires no software installation on the target PC (or device), but I do agree with you that it is a slightly more complicated approach. Yes, I will try to support both ways but the future of the Windows driver depends on Microsoft's future decisions in regards to driver signing and is not under my control.

stereokai, 2018-09-16:
@samartzidis Do you think it will be possible to do usb to usb (instead of bluetooth to usb)

stereokai, 2018-09-16:
>  Why would you prefer the cable connection?

For lower latency :)

RailRanger, 2018-10-22:
I have been using Raspikey for a good while now, it's bug free as far as I can tell. It's a really great solution since you can use it in the BIOS as well. Thanks a lot, I highly recommend you guys try it out if you haven't.

Back to this topic, I saw this http://leshcatlabs.net/2013/09/27/certificate-issues-way-out/comment-page-4/

https://codesigning.ksoftware.net/index_dc.html

Apparently they will sign certificates.

samartzidis, 2018-10-23:
@RailRanger thanks for this. These guys seem to make the signing process simple but the EV certificates that they sell still cost $350/year, plus signing with EV certificates requires cross-signing through Microsoft's dev center portal, which makes the whole thing even more complicated and "costs" time. I am not running a business neither making anything out of WinAppleKey so I really have no intention in getting into these IMO ridiculous complexities. I am glad that WinAppleKey helped so many people out there already but unless something dramatically changes in Microsoft's driver signing policies (which I doubt), I have no intention in pushing this any further (unless I win the lottery and I dont need a real job anymore :P). On the plus side, this whole frustration kicked off the new RaspiKey project, which I think has a MUCH higher potential and use cases than a Windows-only driver solution! Btw, thanks for your feedback on RaspiKey even though it indeed owes most of its stability and performance to the Linux kernel underneath :)

closed 2018-10-23 by samartzidis

TeodorKolev, 2019-01-25:
Thanks for efforts. RaspberryPi solution or any USB seems absurd to me. What is needed is driver. Not hardwere. Can this be transformed to runnable exe or something like that?