samartzidis/WinAppleKey #17: keyboard stalls if fn + shift is pressed a long time

BrainSlayer, 2017-12-26:
if ctrl + fn swap is disabled. the typical page up combination fn + shift + up leads to a total stall of all key pressings.
the system must be rebootet or the driver restartet to get it back to work

Pawel-Dymek, 2018-09-09:
I have problem with Fn key
it is working perfectly... but only once :P 
after using it, it stays pressed until I press Eject (only this key unblock Fn, idk why...)

rodFloress, 2019-07-18:
> 
> 
> I have problem with Fn key
> it is working perfectly... but only once :P
> after using it, it stays pressed until I press Eject (only this key unblock Fn, idk why...)

I have the same problem, any solution or work arround

vipwp, 2019-08-23:
It's a bug, change the code in a1644.c, set  g_FakeFnActive = 0 at the end for the function ProcessA1644Buffer