samartzidis/WinAppleKey #18: working signed driver for apple wired numpad aluminum (A1243)

Beej126, 2018-01-16:
FYI - https://maxtdp.ru/2017/02/07/hidfalum/

samartzidis, 2018-07-01:
Hi. Thanks for the input. I had a quick look at this, first I would be reluctant to install a device driver on my system especially if I wouldn't be able to see the source code and compile it myself!  Secondly, I examined the code signing certificate of the driver binary (.sys) and it does not look like an EV code signing certificate, this is an easy to issue certificate but that also means that this driver wouldnt work on current/latest versions of Windows 10 (x64). Cheers.

closed 2018-07-01 by samartzidis

samartzidis, 2018-07-02:
> It definitely works on latest Win10 x64 (1803). Agreed about taking a
> gamble on unknown driver.

OK, if that's the case then his driver signing certificate was issued
before the strict Microsoft restrictions came out for newly signed drivers.
When his certificate expires he won't be able to just renew it and keep on
signing any new releases.

Beej126, 2018-07-02:
Thanks for explaining. I was wondering. Thankfully it’s perfect as it is. So so grateful it exists at all. I love that wired aluminum keyboard. I get tons of them on eBay for ~$30.

samartzidis, 2018-07-02:
Yeah I agree that this wired keyboard is one of the best ever made by
Apple. What do you do with so many keyboards? :)

Beej126, 2018-07-05:
oh "tons" is an overstatement... i have 5... basically throw one on every machine i sit at for any duration

while i'm waxing on... after the referenced driver does the basics of flipping Command/Windows & Alt, restoring scrl-lock, break, & prtscrn and INSERT!... then F16-F19 above the numpad are a sweet set of empty bonus keys for mapping media functions, etc.

I laser print labels for the replacement key caps and cover them with packing tape for a little longevity