samartzidis/WinAppleKey #20: (not issue) about TESTSIGNING mode

airthomas, 2018-04-11:
Hi,

is need always in TESTSIGNING mode ON to use WinAppleKey? since I switch back to TESTSIGNING mode No after installed. It is not work in my Windows 10.

Thanks, 
Thomas

samartzidis, 2018-04-11:
Hi Thomas. Unfortunately yes, it needs to be on.

closed 2018-04-11 by samartzidis

auwsom, 2018-11-03:
Great piece of software.. How difficult is it to get a signed certificate for the driver? I'm sure you've looked into it but, just checking, as I just installed a driver for the Magic Touchpad 2 from GitHub that was signed, but they may be using the driver from Bootcamp.

samartzidis, 2018-11-04:
Unfortunately it is very difficult (not technically but in terms of the required legal procedures and costs) as Microsoft requires you to set up a registered company and then pay for the expensive certificate. Yes you are 100% correct, you can just pick the bootcamp drivers for Windows and install them natively (people just extract those drivers then and make them available online - this is what you most likely installed). The only problem with that is that the bootcamp driver specifically for the Apple keyboards cannot do the Fn-Ctrl key swapping that my driver does and which is the main reason why I wrote it. I have started working on another project here https://github.com/samartzidis/RaspiKey that aims to drive Apple keyboards completely through hardware. It's still in beta but I am hoping to find some time to finish it soon.