samartzidis/WinAppleKey #21: Any success with Windows 10 1803?

freddellmeister, 2018-06-30:
I upgraded to W10 x64 1803 and my apple keyboard no longer sends the special keys as before.
The machine is still in test mode. And trying to load the device drivers again, nothing seems to happen. In 1709 it was working fine. Did anything change in 1803 that prevents the great tool?

samartzidis, 2018-06-30:
I have noticed this after windows version upgrades. Can you try
reinstalling the driver?

freddellmeister, 2018-06-30:
Hi repairing or reloading did not help, however a full unistall/reinstall made delete work again.
Thanks.
Now CTRL-ALT-DEL does not work though or any of the media keys.
Will reboot later and try again.

samartzidis, 2018-07-01:
I have just updated to the 1803 build and the driver seems to work fine. Like I said, I needed to perform a clean uninstall and reinstall. Let me know if that did n't fix it for you though.

closed 2018-07-01 by samartzidis