samartzidis/WinAppleKey #23: Driver cause BSOD

kleuter, 2018-08-02:
Thank you for your work!.

After I started using the driver, the system (clean Win10 x64, Intel NUC) crashes approx. every 1-2 days. I have memory dumps but there's nothing there related to the driver.

```
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

WHEA_UNCORRECTABLE_ERROR (124)
A fatal hardware error has occurred. Parameter 1 identifies the type of error
source that reported the error. Parameter 2 holds the address of the
WHEA_ERROR_RECORD structure that describes the error conditon.
Arguments:
Arg1: 0000000000000000, Machine Check Exception
Arg2: ffffd289b9bc1028, Address of the WHEA_ERROR_RECORD structure.
Arg3: 00000000b6000000, High order 32-bits of the MCi_STATUS value.
Arg4: 0000000000110135, Low order 32-bits of the MCi_STATUS value.

Debugging Details:
------------------

KEY_VALUES_STRING: 1

STACKHASH_ANALYSIS: 1

TIMELINE_ANALYSIS: 1

DUMP_CLASS: 1

DUMP_QUALIFIER: 401

BUILD_VERSION_STRING:  17134.1.amd64fre.rs4_release.180410-1804

SYSTEM_MANUFACTURER:  Intel Corporation

SYSTEM_PRODUCT_NAME:  NUC7PJYH

SYSTEM_SKU:                                   

SYSTEM_VERSION:  J67992-401

BIOS_VENDOR:  Intel Corp.

BIOS_VERSION:  JYGLKCPX.86A.0041.2018.0626.1500

BIOS_DATE:  06/26/2018

BASEBOARD_MANUFACTURER:  Intel Corporation

BASEBOARD_PRODUCT:  NUC7JYB

BASEBOARD_VERSION:  J67969-401

DUMP_TYPE:  1

BUGCHECK_P1: 0

BUGCHECK_P2: ffffd289b9bc1028

BUGCHECK_P3: b6000000

BUGCHECK_P4: 110135

BUGCHECK_STR:  0x124_GenuineIntel

CPU_COUNT: 4

CPU_MHZ: 5da

CPU_VENDOR:  GenuineIntel

CPU_FAMILY: 6

CPU_MODEL: 7a

CPU_STEPPING: 1

CPU_MICROCODE: 6,7a,1,0 (F,M,S,R)  SIG: 22'00000000 (cache) 22'00000000 (init)

BLACKBOXBSD: 1 (!blackboxbsd)

BLACKBOXPNP: 1 (!blackboxpnp)

DEFAULT_BUCKET_ID:  WIN8_DRIVER_FAULT

PROCESS_NAME:  System

CURRENT_IRQL:  f

ANALYSIS_SESSION_HOST:  DESKTOP

ANALYSIS_SESSION_TIME:  08-02-2018 17:00:30.0504

ANALYSIS_VERSION: 10.0.17712.1000 amd64fre

STACK_TEXT:  
ffff9901`714daae8 fffff802`1b7a1520 : 00000000`00000124 00000000`00000000 ffffd289`b9bc1028 00000000`b6000000 : nt!KeBugCheckEx
ffff9901`714daaf0 fffff800`0a0a14ba : ffffd289`b9bc1028 ffffd289`b87f9440 ffffd289`b87f9440 ffffd289`b87f9440 : hal!HalBugCheckSystem+0xd0
ffff9901`714dab30 fffff802`1b0c9021 : ffffd289`b9bc1028 00000000`00000000 ffffd289`b87f9440 ffffd289`b87f9440 : PSHED!PshedBugCheckSystem+0xa
ffff9901`714dab60 fffff802`1b7a2c60 : 00000000`00000728 00000000`00000003 00000000`00000000 00000000`00000000 : nt!WheaReportHwError+0x261
ffff9901`714dabc0 fffff802`1b7a2fc0 : 00000000`00000010 ffffd289`b87f9490 ffff9901`714dad68 00000000`00000003 : hal!HalpMcaReportError+0x50
ffff9901`714dad10 fffff802`1b7a2eae : ffffd289`b87369a0 00000000`00000001 00000000`00000000 00000000`00000000 : hal!HalpMceHandlerCore+0xe0
ffff9901`714dad60 fffff802`1b7a30f2 : 00000000`00000004 00000000`00000001 00000000`00000000 00000000`00000000 : hal!HalpMceHandler+0xda
ffff9901`714dada0 fffff802`1b7a3390 : ffffd289`b87369a0 ffff9901`714dafd0 00000000`00000000 00000000`00000000 : hal!HalpMceHandlerWithRendezvous+0xce
ffff9901`714dadd0 fffff802`1afa6b3b : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : hal!HalHandleMcheck+0x40
ffff9901`714dae00 fffff802`1afa67ff : 00000000`00000000 fffff802`1afa6717 00000000`00000000 00000000`00000000 : nt!KxMcheckAbort+0x7b
ffff9901`714daf40 fffff802`1b09a80a : ffffc268`b1504000 fffff802`1b09a465 00000000`00000000 fffff802`1b09a522 : nt!KiMcheckAbort+0x1bf
ffffe60e`73029920 fffff802`1b09a00f : ffffc268`b1504000 00001700`3b40b3dd 00000000`00000000 00000000`ffffffff : nt!RtlpGenericRandomPatternWorker+0x27a
ffffe60e`73029970 fffff802`1b07b3a9 : ffffe5e1`3458a820 00000000`00000000 ffffc268`b1504000 00000000`00012151 : nt!RtlScrubMemory+0x113
ffffe60e`730299b0 fffff802`1b07b22b : 00000000`00000001 ffffd289`b863cd40 ffffe780`00bfa180 ffffe60e`00000009 : nt!MiScrubPage+0xe1
ffffe60e`730299f0 fffff802`1b56748f : ffffd289`b95b9c00 fffff802`00000000 ffffd289`b95b9c68 ffffd289`b95b9c68 : nt!MiScrubNode+0x1a7
ffffe60e`73029a80 fffff802`1ae42455 : 00000000`00000300 ffffd289`ba7f2040 ffffd289`b86edaf0 ffffd289`b86edaf0 : nt!MiScrubMemoryWorker+0x6f
ffffe60e`73029b80 fffff802`1aee4d87 : ffffd289`ba7f2040 00000000`00000080 ffffd289`b86db440 ffffd289`ba7f2040 : nt!ExpWorkerThread+0xf5
ffffe60e`73029c10 fffff802`1af9fa06 : fffff802`19c55180 ffffd289`ba7f2040 fffff802`1aee4d40 00000000`00000246 : nt!PspSystemThreadStartup+0x47
ffffe60e`73029c60 00000000`00000000 : ffffe60e`7302a000 ffffe60e`73024000 00000000`00000000 00000000`00000000 : nt!KiStartSystemThread+0x16

THREAD_SHA1_HASH_MOD_FUNC:  0707e8acbf3ce02513bdd7e95a56085c6d3a364a

THREAD_SHA1_HASH_MOD_FUNC_OFFSET:  660459cec8750ea6782e69c4fd3dfbf63b710b85

THREAD_SHA1_HASH_MOD:  af012a6d5c2a26db9fd59ba3248350c1617f6fc5

FOLLOWUP_NAME:  MachineOwner

MODULE_NAME: GenuineIntel

IMAGE_NAME:  GenuineIntel

DEBUG_FLR_IMAGE_TIMESTAMP:  0

STACK_COMMAND:  .thread ; .cxr ; kb

FAILURE_BUCKET_ID:  0x124_GenuineIntel_PROCESSOR_CACHE

BUCKET_ID:  0x124_GenuineIntel_PROCESSOR_CACHE

PRIMARY_PROBLEM_CLASS:  0x124_GenuineIntel_PROCESSOR_CACHE

TARGET_TIME:  2018-07-30T20:28:10.000Z

OSBUILD:  17134

OSSERVICEPACK:  0

SERVICEPACK_NUMBER: 0

OS_REVISION: 0

SUITE_MASK:  272

PRODUCT_TYPE:  1

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

OSEDITION:  Windows 10 WinNt TerminalServer SingleUserTS

OS_LOCALE:  

USER_LCID:  0

OSBUILD_TIMESTAMP:  2018-07-14 05:53:27

BUILDDATESTAMP_STR:  180410-1804

BUILDLAB_STR:  rs4_release

BUILDOSVER_STR:  10.0.17134.1.amd64fre.rs4_release.180410-1804

ANALYSIS_SESSION_ELAPSED_TIME:  e87

ANALYSIS_SOURCE:  KM

FAILURE_ID_HASH_STRING:  km:0x124_genuineintel_processor_cache

FAILURE_ID_HASH:  {4c8f3f5e-1af5-ed8b-df14-d42663b1dfa7}

Followup:     MachineOwner
---------
```

samartzidis, 2018-08-03:
If you still have the memory dump or next time it happens, try to follow this guide to identify the driver (.sys) that is possibly causing this. https://www.silabs.com/community/interface/knowledge-base.entry.html/2014/10/21/how_to_determinewhi-9VIf

kleuter, 2018-08-03:
The post above shows the ouput of the "analyze -v" command, it doesn't give a driver name

samartzidis, 2019-02-15:
> The post above shows the ouput of the "analyze -v" command, it never gives a driver name

It's a bit late already but can you let me know which model of the keyboard were you using?

Thanks.

closed 2020-01-19 by samartzidis