samartzidis/WinAppleKey #28: Setting SwapFnCtrl to 0 Results in 'stuck FN' key

zeus82, 2018-11-21:
If you set SwapFnCtrl to 0 then if you press the 'FN' key the keyboard stops working as expected.

For example, the up arrow stays as page-up, even though I am not pressing the FN key.  unplugging and re-plugging in the keyboard fixes the issue until I press the FN key again.

vogler, 2019-11-07:
I have the same problem. However, for me, re-plugging didn't fix the issue. Had to set it to 1 again with the onscreen keyboard and reboot.

Keyboard 579C-A1644
Windows Version	10.0.18362 Build 18362

reznikartem, 2019-12-28:
Same problem with Apple Magic Keyboard 2

reznikartem, 2019-12-30:
Make PR with fix for this problem

https://github.com/samartzidis/WinAppleKey/pull/39

Attach zip file with Setup binary, if you cant compile it by yourself

[Setup.zip](https://github.com/samartzidis/WinAppleKey/files/4010276/Setup.zip)

samartzidis, 2020-01-18:
Should be addressed now by v1.0.7 thanks to @reznikartem

closed 2020-01-18 by samartzidis