samartzidis/WinAppleKey #30: Cannot get keyboard working after upgrading from 1709 to 1803

idliguy, 2018-12-14:
@samartzidis I've been using this in 3 machines without any issues, but in one of my newer machines I got it working once (Did not reboot immediately after install), installed an update and then rebooted the machine. And now the keyboard works only the old way with this machine alone.

I tried multiple uninstallation, reboot, reintallation and reboot, and the only thing I notice is that no registry entry is created under "Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\" (It creates this path "Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\WinAppleKey" in my other machines, and I tried exporting and importing but that does not help as well). Another thing I noticed was that it would not place inf and .sys files in C:\Windows\INF and C:\WIndows\System32\drivers respectively.

Any help would be much appreciated since this would be the machine that I will be using frequently for the next few months.

Thanks

kranix0, 2019-04-13:
Same here. I went to install the latest WinAppleKey for the first time and found that it didn't work. Then after I saw this report I decided to try 1.0.4, which appears to work fine.

samartzidis, 2019-04-14:
I have identified issues with the .inf based installer on latest builds of Windows 10 (1803+). I will work on it the soonest I have some spare time but in order to protect users from trouble I have taken all the current builds offline. Thanks.

samartzidis, 2019-04-23:
There is a new release 1.0.6 that could potentially fix your problem.

closed 2019-06-08 by samartzidis