samartzidis/WinAppleKey #33: Use self-signed certificate?

kranix0, 2019-06-10:
Has anyone had success with self-signing the driver? Is this a viable workaround for the lack of an officially-signed driver?

kranix0, 2019-06-10:
Ah… I see that the driver is already self-signed. Is there a workaround so that I don't need to be in TESTSIGNING mode all the time? I tried turning it off but then the driver stopped working.

samartzidis, 2019-06-10:
Hi @kranix0, yes the driver is already signed with a "dummy" test certificate as required by Windows to install. The purpose of driver signing is to establish the authenticity of a company that sells a device driver; and the integrity of the driver binary itself. Consumers are not expected to sign any device drivers whatsoever.

Developing device drivers for Windows does not play well with *free* and *open source*. So unfortunately there is no way to get around the TESTSIGNING requirement, unless someone sets up a registered software company and starts selling the driver as a commercial product, provided they have also purchased an expensive driver signing certificate by Microsoft or a "reseller".  I know this is an issue for some users but I believe it is better than nothing.

closed 2019-06-10 by samartzidis

kranix0, 2019-06-10:
Agreed, definitely much better than nothing. It's a fantastic driver and I appreciate your work! It's great to see more free sofware on Windows. I just wish that Microsoft didn't make it so difficult for projects like this.