samartzidis/WinAppleKey #35: Registry keeps resetting to default values

brhsiao, 2019-06-20:
SwapAltCmd and SwapFnCtrl keep on resetting to the default values (0 and 1 respectively) without my interference. I'm not sure when this is happening -- on restart? on keyboard reconnect?

brhsiao, 2019-06-22:
I figured out when it happens. If I switch from USB to Bluebooth or vice versa  the settings reset to their default.

When does this happen exactly? I looked at the code and it only ever reads from the registry, not writes.

lukiolee, 2019-06-22:
Yeah i have the same problem bro

reznikartem, 2019-12-30:
Make PR with fix for this problem

https://github.com/samartzidis/WinAppleKey/pull/39

Attach zip file with Setup binary, if you cant compile it by yourself

[Setup.zip](https://github.com/samartzidis/WinAppleKey/files/4010276/Setup.zip)

samartzidis, 2020-01-19:
Should be now fixed in 1.0.7

closed 2020-01-19 by samartzidis