samartzidis/WinAppleKey #36: Custom built driver not getting recognized

sonictk, 2019-07-11:
I tried compiling the solution along with making the installer WiX project. When I use the installer built from source, everything builds and installs successfully, but Windows never uses the built driver instead, always favoring the one built into Windows. When I use the installer provided from the releases page, everything works fine. 

Is there something that I need to do during compilation and building the installer that I forgot? I can't actually tell why Windows keeps favoring the built-in driver instead of the one I built from source instead.

vipwp, 2019-08-20:
I got the same issue.

vipwp, 2019-08-23:
> I tried compiling the solution along with making the installer WiX project. When I use the installer built from source, everything builds and installs successfully, but Windows never uses the built [...]

I installed the release setup.msi,then relace the c:\\windows\system32\drivers\WinAppleKey.sys with my own build one,then reboot. It works on my OS(win10 x64 build 15063 release 170317-1834).

sonictk, 2019-10-04:
Hey, @vipwp thanks for the tip! I'll give that a go.

samartzidis, 2020-01-29:
That should be fixed now in v1.0.7.

closed 2020-01-29 by samartzidis