samartzidis/WinAppleKey #41: WinAppleKey 1.0.6 doesn't work

Haikgh, 2020-01-16:
Hi,
As the tile says. 
I have enabled the testers mode and installed the Setup.msi, however it doesn't seem to  install properly as there is no trace of this driver in the Registry.
I understand it is not just me who has issues with the  V 1.0.6. Could you please make the 1.0.4 and 1.0.5 available for download?
Thanks

wjdghks7600, 2020-01-18:
mine neither maybe because of the update of bootcamp driver

samartzidis, 2020-01-18:
I just released a 1.0.7 build. It has been tested against a vanilla Windows 10 machine and the driver installs and works fine. You may want to give this a go? To your comment, some previous releases were buggy in other areas (potential BSOD) and I explicitly removed them. 
As an extra note, on a normal installation you should be getting a popup such as this:
[image]
where you should click **Install**.

wjdghks7600, 2020-01-22:
cool it works well I appreciate you again

Haikgh, 2020-01-22:
> I just released a 1.0.7 build. It has been tested against a vanilla Windows 10 machine and the driver installs and works fine. You may want to give this a go? To your comment, some previous releases [...]

Hi,
I just don't get that driver installation screen during the installation of teh Setup. Is there something I am missing?

Thanks

samartzidis, 2020-01-22:
@Haikgh that would be the case if you installed before with the checkbox checked I guess.
Does this version work for you now or still having issues?

Haikgh, 2020-01-22:
> @Haikgh that would be the case if you installed before with the checkbox checked I guess.
> Does this version work for you now or still having issues?

Yes, the driver installation message has appeared the very first time I have installed the version 1.0.6.
No, the WinAppleKey still doesn't work, as in it's nowhere to be found in my registry even after installing the version 1.0.7. 
How else can I verify if WinAppleKey has been installed properly? Am I supposed to see a new driver in my Device Manager ? Because I don't seem to see anything there from WinAppleKey.

samartzidis, 2020-01-22:
One thing that I suspect is that the installer does not detect your keyboard hardware id as expected. Can you open device manager/keyboards and get a window screen shot of the **Magic Keyboard** device's "Hardware Ids" property? It should be a list containing several HID\VID... entries.

Haikgh, 2020-01-23:
Hello,
Here is what I've got in the Device Manager. AS you can see there is nothing that has any mention of Magic Keyboard or Winapplekey.
I have also checked those two locations for the driver and couldn't find any file which would contain the name winapplekey. As if i have never even installed the version 1.0.7 Setup.msi.
The installation process goes without any errors, tho, which is why it is so confusing that I can not locate any trace of the Winapplekey anywhere on my PC. I have even searched the entire Registry, and no mention there, either.
[image]
[image]
Any idea why this might be the case ?
Thank you

samartzidis, 2020-01-23:
@Haikgh can you open the "Hardware IDs" property of the "HID Keyboard Device" that shows up when you connect your Apple keyboard and see what's there under "Value"? You may need to check both entries that show up in there to find which one is which. Also use e-mail please. Thanks.

closed 2020-01-23 by samartzidis

Haikgh, 2020-01-23:
> @Haikgh can you open the "Hardware IDs" property of the "HID Keyboard Device" that shows up when you connect your Apple keyboard and see what's there under "Value"? You may need to check both entries that show up in there to find which one is which. Also send by e-mail instead please. Thanks.

Email has been sent.
Thank you

samartzidis, 2020-01-23:
As a follow-up comment, support for the old A1314 model (as was in this case) was removed after v1.0.4 due to a potential BSOD bug. Nevertheless, I just re-uploaded the old v1.0.4 release due to several requests. Use at your own risk.