samartzidis/WinAppleKey #48: Huge Thanks!

HouzuoGuo, 2020-05-09:
My colleagues were complaining that my mechanical keyboard was too loud over microphone, actually I didn't like the keyboard that much anyways as the keys are too tall for my taste.

I was a Macbook user about a year ago and really loved the keyboard. Couple of hours ago I scavaged this slightly used, good as new apple magic keyboard from miscellaneous section of local electronics store, was happy to bring it back home, but soon realised that it doesn't even have a page up and down key,

Gladly I found your project :) and it's working beautifully.

HouzuoGuo, 2020-05-09:
BTW - this is kinda weird. Ctrl+LShift+T doesn't work in Chrome and Edge, though Ctrl+RShift+T works.

Ctrl+LShift+Delete(eject) works though.

samartzidis, 2020-05-10:
Thank you @HouzuoGuo :) I am aware of the Ctrl(fn)+LShift+T thing. Strangely it seems to be an issue with the keyboard hardware itself not liking this particular combination.

HouzuoGuo, 2020-05-12:
decades long rivalry between PC and MAC haha.. thanks!

closed 2020-05-12 by HouzuoGuo