samartzidis/WinAppleKey #49: Ctrl+LShift+T works (with the "ctrl" key instead of "fn")!

HouzuoGuo, 2020-05-29:
Hey!

Should have read your manual better. Couple of weeks into using your software, after having already  developed muscle memory of using Ctrl(FN)+RShift+T to restore the last tab, I got the curiosity of leaving FN and Ctrl keys in-place without swapping them through registry.

And it Ctrl+LShift+T works again!

Now I just have to retrain muscle memory and forget Ctrl+RShift+T...

joowani, 2020-07-26:
It's not just ctrl + lshift + T. When FN and Ctrl keys are swapped, Ctrl modifier is not registered properly in many cases (e.g. chrome, logitech gaming software). I had to swap them back to get things to work correctly. I believe this is a bug.

HouzuoGuo, 2020-07-27:
The author of this program clarified that FN key may be having a hardware issue - https://github.com/samartzidis/WinAppleKey/issues/48

samartzidis, 2020-10-22:
https://github.com/samartzidis/WinAppleKey#3-when-running-in-the-default-fn-ctrl-swap-mode-the-ctrl-lshift-t-combination-commonly-used-to-reopen-a-closed-browser-tab-in-chrome-does-not-work

It is a hardware limitation of the keyboard.

closed 2020-10-22 by samartzidis