samartzidis/WinAppleKey #58: Didn't work on windows 11？

o000u, 2021-08-18:
Since I upgraded my system to windows 11, I found that it didn't work. Does anyone have the same problem?

samartzidis, 2021-08-21:
It won't work if Secure Boot is enabled in the BIOS and Windows is not in Test Mode, as mentioned in the README. Other than that I did n't have the chance to try Windows 11 yet.

o000u, 2021-09-06:
Before, it worked perfectly under Windows 10. Later, it didn't work when I upgraded to windows 11. My BIOS settings remained unchanged～～～

vxsl, 2023-04-14:
It's also not working for me on Windows 11. I haven't had the chance to try it on Windows 10, but I've enabled TESTSIGNING mode and disabled secure boot, so I suspect win11 is likely the problem.

apppps, 2023-06-26:
need to update

Kiswelrg, 2023-10-06:
guys does windows11 work now?