samartzidis/WinAppleKey #62: Requires support for use on Windows 11.

apppps, 2023-06-19:
Hi there.
I can't swap fn key and control key in windows 11.
Apple Magic Keyboard 2 is a great product and needs to be supported on Windows 11. Thank you for the quick fix
Thank you.

jianguyuxing, 2023-06-26:
I use same Keyboard on Win 11, and I can swap fn and ctrl key.

apppps, 2023-06-26:
> I use same Keyboard on Win 11, and I can swap fn and ctrl key.

I went through all the steps and it's win11 home, but still winapplekey doesn't work. can you tell me what steps you went through, maybe win11 pro?

jianguyuxing, 2023-07-01:
> > I use same Keyboard on Win 11, and I can swap fn and ctrl key.
> 
> I went through all the steps and it's win11 , keyboard(A1843) but still winapplekey doesn't work. can you tell me what steps you?

oh,mine is A1644. I finished all procedure which described in readme, then reboot computer.

apppps, 2023-07-01:
> > > Win 11에서 동일한 키보드를 사용하고 fn과 ctrl 키를 바꿀 수 있습니다.
> > 
> > 
> > 나는 모든 단계를 거쳐 win11 , keyboard(A1843) 이지만 여전히 winapplekey 가 작동하지 않습니다. 당신이 어떤 단계를 말해 줄 수 있습니까?
> 
> 오, 제 것은 A1644입니다. readme에 설명된 모든 절차를 완료한 다음 컴퓨터를 재부팅합니다.

 A1644 I've tested it, but it doesn't work.
Are you sure it's Windows 11?